openrelics/enigma #12: NBP Centering Device 100039 missing thickness dimensions

billstrong, 2020-10-05:
There are not thickness dimensions on either the plate or hub of 100039 centering device.

mariusstrom, 2020-10-05:
Note that if direct measurements are taken, Section A is a 2:1 magnification so ensure that measurements are relative to the ones drawn.

I don't recall how I picked up dimensions of this one as I modeled it. This model is also drawn oddly - check this out:

[image]

Why the extra line beyond the R5 circle?

billstrong, 2020-10-06:
To answer your question about the fillet, I think they are just calling out that the corner arc is a fillet constructed atop that hard edge geometry.