openrelics/enigma #19: NBP 400006 Clamp has missing fillet dimensions

billstrong, 2020-10-06:
The drawing for this part calls out a region of free stock flow between two planar features in sheet metal. The geometry shown, however, shows the transition radii between the planar and free flow regions as protruding into the planar region, without specifying this radii.

Fix may become apparent in assembly.

mariusstrom, 2020-10-06:
I'm not sure that needs to be modeled - it's a spring steel and the notches to bend the spring steel are modeled (unless I'm misinterpreting your mechanical engineering speak, which is quite possible!)

billstrong, 2020-10-06:
No, I tend to agree with you. Just documenting it since it is technically a deviation in our model

closed 2020-10-06 by mariusstrom