openrelics/enigma #24: Rename Fusion 360 project from "Full-Size Enigma Machine" to "openrelics-enigma-m1" or similar?

billstrong, 2020-10-07:
A minor thing... I manage a number of Fusion 360 projects, and it is nice to have some context in the file listing. I propose we rename the project to include the text "OpenRelics" (or "open-relics", or "openrelics") or some such tag so it is easily identifiable. Further, the project name should identify the machine is an M1 or an Enigma I variant, if we are certain of the precise designation of the machine we are trying to replicate.

mariusstrom, 2020-10-07:
Yes, this is definitely on the todo list. We should come up with an openrelics project naming taxonomy, but I'm ok with openrelics-engima-m1 for now.. although we could model M4 parts in the same project, there's a lot of reuse, just the assemblies are different and a few new components

billstrong, 2020-10-07:
I'd be inclined toward independent PNs and models, but with a system of references for identical parts between projects. Such as solution could accommodate very slight differences in otherwise identical parts. The exception to this is modern and period COTS items which are confirmed to be identical; these I think ought to reside in their own projects in the long term.

mariusstrom, 2020-10-07:
oh, that's an interesting notion. Can Fusion assemblies pull from documents across projects? That would be awesome if so (if not a permissions nightmare, but that's ... managable)

mariusstrom, 2020-10-08:
Completed and renamed as openrelics-enigma-m1

closed 2020-10-08 by mariusstrom