openrelics/enigma #29: NBP 100116 Baseplate print appears to be missing reference to 100023

billstrong, 2020-10-08:
Issue is resolved in 3D model for 100116 but not discrepancy is not documented elsewhere.

mariusstrom, 2020-10-08:
Are you referring to the NBP parts BOM missing 100023?

billstrong, 2020-10-08:
Yes, that's right

mariusstrom, 2020-10-08:
interesting. I notice that 100021 and 100023 have identical length dimensions. I wonder if they referenced the wrong parts?

However, 100021 has M2 screw holes as opposed to M4 on 100023. 

We should check which parts mate to them to figure out which goes where.

LukePeterson, 2021-10-12:
I believe the 100116 drawing is correct - it's just missing the 100023 reference as @billstrong has mentioned.  100023 mates with 100020 (contactor plate) which, along with the baseplate, also has its own 4.2 diameter screw hole.  There's only one instance of 100023 in the machine from what I can tell. 

Unreferenced 100023 parts are highlighted in yellow (the screw is referenced, but highlighted for completeness):

[image]