openrelics/enigma #31: Springs and pins in 400201 Roll IV Assy are not fully constrained

billstrong, 2020-10-09:
In the local assy, this is hard to catch. Once the rotor is in a higher level assy, rotating it causes the pins to jog out of place. Likely need to change a constraint from cylindrical to slider.

mariusstrom, 2020-10-09:
did you circular pattern the pins? circular pattern does _not_ copy joints which is really irritating.

mariusstrom, 2020-10-09:
(the reflector assembly suffers from the same on the Bushing and Blanks)

billstrong, 2020-10-09:
I did circular pattern; maybe we should just bite the bullet and do individual joints? For patterns I have been creating a rigid group.

mariusstrom, 2020-10-09:
ah, I didn't know about rigid groups. 👍 

I think we need to decide if individual joints or group are the right things. For spring pins, theoretically they can move independently based on pressure to the next contact die over, but in practice, they should all move in an identical motion at the same time. For rotational and translation motions of the reflector and rotor, it's a fixed system so a rigid group should be fine.

billstrong, 2020-10-09:
I think the joints should reflect reality. So, when they are rigid, patterns are okay; can always add a redundant joint on one or two of the patterned parts to make sure the pattern is correct. For the spring pins, they really should all be able to move independently. Practically it is useful to be able to do this for fit checking and such.

billstrong, 2020-10-12:
Fixed in rotor and reflector.

closed 2020-10-12 by billstrong